fraschetti/Octoslack #88: ERROR - Slack API message send error: invalid_arguments

tedder, 2020-05-07:
Looks like after upgrading to a "new bot" token in #86, messages fail:

```
2020-05-07 17:43:07,023 - octoprint.plugins.Octoslack - ERROR - Slack API message send error: invalid_arguments
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_Octoslack/__init__.py", line 3048, in send_slack_message
    icon_emoji=slack_icon_emoji,
  File "/home/pi/oprint/local/lib/python2.7/site-packages/slacker/__init__.py", line 381, in post_message
    'reply_broadcast': reply_broadcast
  File "/home/pi/oprint/local/lib/python2.7/site-packages/slacker/__init__.py", line 123, in post
    api, **kwargs
  File "/home/pi/oprint/local/lib/python2.7/site-packages/slacker/__init__.py", line 99, in _request
    raise Error(response.error)
Error: invalid_arguments
```

This is certainly opaque, but [the `as_user` entry](https://api.slack.com/methods/chat.postMessage#arg_as_user) is certainly a problem: "[as_user] may not be used with newer bot tokens."

I'm testing locally to see if that's the only problem.

tedder, 2020-05-07:
FTR I have `slacker` version 0.12 installed, the setup.py here doesn't restrict a version, so I guess I'll test against 0.12 and also 0.14 (latest).

fraschetti, 2020-05-08:
Hi @tedder 

You're right on my heels with your analyis and changes.
I had been debating whether or not to remove this functionality completely or keep backwards compatibility for folks with older bot tokens.
My current path is to allow backwards compatibility for a but to let people know Slack has deprecated support for these options and they should expect them to be removed from Octoslack in a future release.
I've finished the Python side of these changes and am now working on my least favorite part - the config UI.
For completeness, the **as_user** argument as well as **icon_emoji** and **icon_url** are no longer supported for new bot tokens.

Once I get the UI wrapped up, I'll run through my usual battery of tests and then get this thing out in the wild.

tedder, 2020-05-08:
Heh, yeah, I dislike UI too :) I definitely think the options merit removal in the future, but you have a good handle on it.

fraschetti, 2020-05-24:
Hi @tedder 

The new 2.0.0 release should address this - the plugin will default to not support setting those values (as they're not supported and providing them will result in an error) but I've added Classic bot support for folks to maintain access to these customizations for folks still on the old bots.

closed 2020-05-24 by fraschetti