fraschetti/Octoslack #90: Errors cause print to cancel when snapshot stream missing

nebbywan, 2020-05-14:
I had to remove a camera from my printer setup to help deal with the consequences of the rona, and the resulting errors from this missing stream caused the print to cancel as soon as the "print started" slack message was sent. Removing this stream from the configuration worked as a workaround.

I'm not sure if this would impact the scenario at all, but I'm actually running Octoprint on a Ubuntu 20.04 box, the config is different than the common OctoPi setup and this has caused one dependency issue in the past.

Here is the offending error from the Octoprint log: 
[Octoslack.txt](https://github.com/fraschetti/Octoslack/files/4630999/Octoslack.txt)

Please let me know if there is any more info you need and thanks for your work on this plugin.

fraschetti, 2020-05-17:
Hi @nebbywan 

It looks like you have a handful of plugins all in the mix. Do you mind doing a couple of test runs to confirm the issue?
1. Problematic camera enabled + All plugins **except** Octoslack enabled
2. Problematic camera enabled + **Only** Octoslack enabled

Octoslack itself wouldn't ever cancel a print unless it received a Slack message to do so. OctoPrint itself might have some internal scenarios where it deems it necessary to cancel a print but the attached log doesn't shed any light on that (we only turned on debug logging for Octoslack, not OctoPrint), or perhaps another plugin is at fault. I've put all of Octoslack's snapshot image fetch into background threads to prevent them from impacting the print / other plugins but you never know.
Please let me know your tests go and we'll take it from there.

Thanks in advance!

nebbywan, 2020-05-18:
Hmm. #1 was how I got it to work last time after ruling out a couple other plugins, so I went straight to #2 and it worked fine as well. Then I tried all plugins enabled and everything continued to work fine despite still seeing the same errors in the log as before. Must be a quirk, so I'll close this for now and if I pops up again I'll come back to it.

Thanks!

closed 2020-05-18 by nebbywan