fraschetti/Octoslack #95: Amazon S3 - HTTPError: 400 Client Error

SebSeifert, 2020-07-09:
Hi,
first thanks for this awesome Addon.
But i'm not able to Upload Images to S3. MinIO is working fine.
First i had the Problem with the 403 Error. I could see that in the AWS Console the new generated root Key or IAM Key are never used. Now suddenly it seems to be fixed. BUT now ich get the 400 Error. 
octoprint.log:
`2020-07-09 13:17:42,596 - octoprint.plugins.Octoslack - ERROR - Failed to upload asset to S3: 400 Client Error: Bad Request for url: https://derseboctoprint.s3.amazonaws.com/Snapshot_30351ed6c1de11ea8649b827ebd486a4.png
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_Octoslack/__init__.py", line 4685, in upload_asset
    expires=s3_expiration,
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tinys3/connection.py", line 171, in upload
    return self.run(r)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tinys3/connection.py", line 262, in run
    return self._handle_request(request)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tinys3/connection.py", line 356, in _handle_request
    return request.run()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tinys3/request_factory.py", line 347, in run
    r.raise_for_status()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/models.py", line 940, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
HTTPError: 400 Client Error: Bad Request for url: https://derseboctoprint.s3.amazonaws.com/Snapshot_30351ed6c1de11ea8649b827ebd486a4.png`

May somebody can support me with the AWS Access System/Controls?

jonkan, 2020-11-13:
I think I'm having the issue.. The log just says `ERROR - Failed to upload asset to S3: 403 Client Error: Forbidden for url: https://<redacted-bucket-name>.s3.eu-north-1.amazonaws.com/Snapshot_386ff568260511ebbe7adca6322e8e46.png`
I've verified the access token via the aws cli.. 
Did you find a solution @SebSeifert?

fraschetti, 2020-11-20:
Hi @DerSeb90 and @jonkan 

Unfortunately the HTTP 400 and 403 alone don't indicate exactly why S3 isn't happy with your request.
I suspect you both have different issues but luckily the next step is the same for both of you... enable debug logging for the plugin so we can take a look at the payload the S3 endpoint is sending back.

Do you mind providing some rough details about your environment?
-What device/OS are you running?
-Do you mind following the steps below to generate a debug log for analysis?

1. Enable DEBUG logging
2. Reproduce the issue
3. Send Octoprint.log

Instructions to enable DEBUG logging for Octoslack

* Enable DEBUG logging
Ensure the following has been added to /home/pi/.octoprint/logging.yaml  (or appropriate path for your OctoPrint installation)
```
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
```

* Restart OctoPrint
```
sudo service octoprint restart
```

For OctoPi releases, the log file will be located in:
/home/pi/.octoprint/logs/octoprint.log

jonkan, 2020-11-27:
@fraschetti Thanks for your response, sorry for the delay. 
This is my log from sending a `@<bot> status` message:
```
2020-11-27 10:13:05,554 - octoprint.plugins.Octoslack - DEBUG - Sending Slack reaction - Channel: C01ER3JAWAW, Timestamp: 1606471983.005700, Reaction: stopwatch, Remove: True
2020-11-27 10:13:05,948 - octoprint.plugins.Octoslack - DEBUG - Successfully sent Slack reaction - Channel: C01ER3JAWAW, Timestamp: 1606471983.005700, Reaction: stopwatch, Remove: True
2020-11-27 10:13:06,271 - octoprint.plugins.Octoslack - ERROR - Failed to upload asset to S3: 403 Client Error: Forbidden for url: https://home-printy.s3.eu-north-1.amazonaws.com/Snapshot_23b77fee309911eb8145dca6322e8e46.png
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_Octoslack/__init__.py", line 4685, in upload_asset
    expires=s3_expiration,
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tinys3/connection.py", line 171, in upload
    return self.run(r)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tinys3/connection.py", line 262, in run
    return self._handle_request(request)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tinys3/connection.py", line 356, in _handle_request
    return request.run()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tinys3/request_factory.py", line 347, in run
    r.raise_for_status()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/models.py", line 943, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
HTTPError: 403 Client Error: Forbidden for url: https://home-printy.s3.eu-north-1.amazonaws.com/Snapshot_23b77fee309911eb8145dca6322e8e46.png
2020-11-27 10:13:06,273 - octoprint.plugins.Octoslack - DEBUG - Deleting local asset after upload: /tmp/Snapshot_23b77fee309911eb8145dca6322e8e46.png
2020-11-27 10:13:06,275 - octoprint.plugins.Octoslack - DEBUG - Deleting local file (attempt #1): /tmp/Snapshot_23b77fee309911eb8145dca6322e8e46.png
2020-11-27 10:13:06,277 - octoprint.plugins.Octoslack - DEBUG - Deletion of local file confirmed (attempt #1): /tmp/Snapshot_23b77fee309911eb8145dca6322e8e46.png
```
Hmm ok, it works if using a `us-east-1` bucket, as mentioned in https://github.com/fraschetti/Octoslack/issues/102#issuecomment-709839697. Seems we need a way to config a different endpoint url for tinys3, as specified in [Specifying a different endpoint](https://github.com/smore-inc/tinys3), to support other bucket regions..

fraschetti, 2022-11-18:
Queued up in the development branch is a change from the tinys3 library to Amazon's official boto3 library. With this will come support for buckets in other regions. This change will be included in the next release.

zchryr, 2022-12-19:
@fraschetti Any updates on this? I'm running into similar issues :/