fraschetti/Octoslack #98: Pushover "Print finished" notifications don't have File and Estimated Print Time

sid3windr, 2020-08-27:
While the "Print started" notification does contain this info correctly, the "File" and "Estimated print time" on the "Print finished" notification just show "N/A".

Not sure if this is specific to Pushover, but it's what I am using.
I tried looking at the code through GitHub but it looks like the issue shouldn't be Pushover-specific...

Running Octoslack 2.0.0.

fraschetti, 2020-11-20:
Hi @sid3windr 

I can't imagine there's anything specific to Pushover but you never know.

Do you mind providing some rough details about your environment?
-What device/OS are you running?
-How are you connecting to Slack (Webhook or API token)?
-Do you mind following the steps below to generate a debug log for analysis?

1. Enable DEBUG logging
2. Reproduce the issue
3. Send Octoprint.log

Instructions to enable DEBUG logging for Octoslack

* Enable DEBUG logging
Ensure the following has been added to /home/pi/.octoprint/logging.yaml  (or appropriate path for your OctoPrint installation)
```
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
```

* Restart OctoPrint
```
sudo service octoprint restart
```

For OctoPi releases, the log file will be located in:
/home/pi/.octoprint/logs/octoprint.log

sid3windr, 2020-12-12:
Hi,

Apologies for the late reply - I got the e-mail then got quite busy...

I'm not using Slack notification, only Pushover, on a Raspberry Pi with Raspbian/Debian 10 (through OctoPi). Updated to 1.5.1 in the meanwhile.

I have enabled debug logging and will feed back after my next print. Apologies for still being a bit vague.

Thanks!

sid3windr, 2021-05-16:
Once again my apologies for the late reply. I did enable debug logging, didn't find anything in there and then neglected posting it here as well. However I am happy to report after upgrading to both OctoPrint 1.6.1 and the last OctoSlack version yesterday, my Pushover notifications are now correctly reporting filename and print time in the "Print finished" notification!

sid3windr, 2021-05-18:
Hm. Too soon - half of the prints still say n/a !