fraschetti/Octoslack #100: Emojis in Pushbullet

Crowlord, 2020-10-07:
Just jumped on this as telegram plugin is essentially dead.

It integrated with pushbullet just great but my emojis are just showing as text (both on phone and in push). Is there a library i need to include or something or is this just unsupported here? Telegram emojis show up just fine in Pushbullet.

fraschetti, 2020-11-20:
Hi @Crowlord 

This seems like a relatively straightforward issue. Do you mind sharing one of your message configs (where I can see the emoji you're using) and also a screenshot of what you see in Pushbullet? I mainly just want to confirm I see the same thing on my end when I work to fix it.

Thanks in advance.

Crowlord, 2020-12-02:
Sorry only just seen this!

I switched over to Discord instead which works great. 

I will put PB back on my test bed and get back to you.

Crowlord, 2020-12-02:
[image]
[image]

Links2004, 2021-01-10:
looks like matrix has the same problem:

[image]

do you like a new Issue for this or is it the same?

site node:
matrix handles emotes via utf8 or html encode (example bell: `&#x1F514;`) 🔔

Links2004, 2021-01-10:
may https://pypi.org/project/emoji/ helps here.
but no idea in which place in the code its needed.

```py
>> import emoji
>> print(emoji.emojize('Python is :thumbsup:', use_aliases=True))
Python is 👍
```