fraschetti/Octoslack #102: How do snapshots work?

rgordon, 2020-10-16:
I thought that the plugin could capture a snapshot from the camera and send it along with the slack notification upon completion. I first tried just turning the snapshot checkbox on - print completed but no snapshot. 
Then I thought that adding the local camera's URL in the snapshot URL might be the trick - that maybe it just needed to know how to grab the image from the camera - so I tried add this "http://localhost:8080/?action=snapshot" but - that didn't work either. 
My next option is, I guess, the AWS S3 option. Do I presume correctly that the plugin would write the image to S3 and then include the URL for it in the notice - meaning that i'll need public read policies on the bucket? Does it need to list the bucket? Does it need to be setup as a full static web site per AWS?

fraschetti, 2020-10-16:
Hi @rgordon 

There are two ways to capture and send along your snapshots to Slack:
1. Use a Slack bot/application token and not a webhook. 
Assuming you grant the application sufficient permissions, the plugin should be able to upload directly to Slack without a using s hosting service like Imgur or AWS S3.

2. Use a hosting service like Imgur or AWS S3
I personally prefer S3 b/c if configured correctly, it can be free or literal pennies.
Assuming you're using S3, the credentials you supply will of course need access to upload to the S3 bucket but the plugin doesn't make any attempts to list, delete, etc. - it only uploads new files. The files themselves do of course need to be public as when you include a URL in the Slack message, Slack itself fetches that image and then shares a rehosted version for you within your Slack client.

Feel free to give the above a read and then let me know if you have any additional questions. If you're not able to get S3 working, don't stress - there's not much to that configuration and I can whip up a few screenshots for you.

Good luck!

rgordon, 2020-10-16:
Thanks. I was reading the code and experimenting further.
That bot/application token, instead of a web hook url, is probably a better way to go in the long run, soon as i figure all that stuff out more thoroughly. I kind of did the minimal stuff on the Slack side - creating an application, generating a token, etc. but it's using the web hook url interface.

meanwhile, i know the AWS side better (I'm much more of a server-side dev)
I created a bucket, policy, group, user, generated credentials, and had a successful test run at least. so it works.
i did discover that: bucket needs to be in Region us-east-1 (to use that access point you have in the code, no big deal.)
I'll look at further bucket policy restrictions - but it's public readable now anyway - i just don't like giving it more permissions than it needs.

rgordon, 2020-10-16:
yes, thanks, S3 works fine with PutObject and the bucket as a resource. So that's good at least.
being able to specify a prefix "directory" would be nice, because then I could push multiple printers and maybe display them elsewhere too. (last image from printer X kind of thing).
they're small files, i can live with them on the other side of the continent.

and i'll figure out the app/token side of things one of these nights.

closed 2020-10-16 by rgordon