fraschetti/Octoslack #103: Sending message over discord on Windows Server error

norton-chris, 2020-11-12:
I am running this on a Windows Octopi Server. Getting this error on 2 different computers. I have octoslack working on my raspberry pi 3b, seems like it might be a windows server problem, not sure though. Here is the error:
octoprint.plugins.Octoslack - ERROR - Error processing event: Disconnected, Error: 'ValueError' object has no attribute 'message'
Traceback (most recent call last):
  File "c:\users\maker\.conda\envs\octoprint\lib\site-packages\sarge\__init__.py", line 655, in run
    self.process = p = Popen(self.args, **self.kwargs)
  File "c:\users\maker\.conda\envs\octoprint\lib\subprocess.py", line 947, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "c:\users\maker\.conda\envs\octoprint\lib\subprocess.py", line 1416, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

fraschetti, 2020-11-20:
Hi @norton-chris 

Thanks for reporting this issue. Do you mind bumping up the Octoslack log level and sending along an Octoprint.log that reproduces this issue?

* Enable DEBUG logging
Ensure the following has been added to /.octoprint/logging.yaml  (or appropriate path for your OctoPrint installation)
```
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
```

* Restart OctoPrint

* Reproduce issue and send along octoprint.log

norton-chris, 2020-11-21:
Hi @fraschetti, the error was on a university computer and now we're in lockdown. I will get those logs, whenever things open up again.