fraschetti/Octoslack #108: Can't install on new Octoprint 1.5.2

SharpSh00ter109, 2020-12-12:
I'm having trouble installing onto the newest Octoprint version. Install runs and shows completed, but shows unknown plugin. After a reboot plugin isn't installed and is still on the available to install list. Also tried with a fresh install, will not install on any of the Octoprint instances I have. Is this a new issue on the latest version of Octoprint?

vsigler, 2020-12-13:
Just came here to report the very same issue. Plugin installation completes, but after restart it is nowhere to be found, only in the plugin manager as available to install.

fraschetti, 2020-12-14:
Hi @SharpSh00ter109 and @vsigler 

Would you mind restarting OctoPrint and then passing along your octoprint.log file? If the plugin was somehow installed but was failing to load, the interesting bigs would be logged there when OctoPrint starts.

It would also be helpful to see your plugin_pluginmanager_console.log for a day where you've attempted to install Octoslack.

If you're using the OctoPi distribution, both will be located in /home/pi/.octoprint/logs/  - otherwise you'll need to tack these down wherever you've installed Octoprint.

SharpSh00ter109, 2020-12-14:
This is what happens at the end of installing
[image]

And here is each log, I believe these are what you needed
[octoprint.log.txt](https://github.com/fraschetti/Octoslack/files/5685877/octoprint.log.txt)
[plugin_pluginmanager_console.log.txt](https://github.com/fraschetti/Octoslack/files/5685879/plugin_pluginmanager_console.log.txt)

Snortimer, 2020-12-20:
I, too, have this problem. I was trying to replace the Slack plugin as it appears to be unsupported at this point, and exactly the same thing happened to me. I did uninstall and clean out the Slack plugin (and reboot) before installing the Octoslack plugin.

fraschetti, 2020-12-21:
Hi @SharpSh00ter109 @vsigler and @Snortimer 

The issue here is with the official Slack library + Python 2. They abandoned their Python 2 version in a bit of an odd state in that it has a very specific requirement on the version of the Python websocket library. Without forcing things on my side, Python will naturally pick up the latest and greatest websocket library which the Slack client does not like and that causes an error when OctoPrint is installing the prereqs for the Octoslack.
I have a potential fix that I'm testing now - I'm mainly checking to confirm it doesn't cause an issue with OctoPrint itself as we share Python libraries.

Stay tuned and thank you for your patience.

jhubbardsf, 2020-12-24:
@fraschetti Any updates? Or possibly anything one of us can do to help?

fraschetti, 2020-12-24:
Hi @SharpSh00ter109 @vsigler @Snortimer and @jhubbardsf 

I just pushed Octoslack 2.1.0 which should address this issue. Please try your installations again and let me know how it goes.

Unfortunately the simpilest "fix" (forcing the websocket library to match the Slack library requirement) would have introduced a conflict with OctoPrint itself. I had to resort to dropping usage of the Slack RTM API (which is going away so that wasn't a huge deal) and then fix the Slack client API version (to before it introduced the websocket client library version requirement). Luckily all the changes we're missing out on are related to the RTM APIs which as stated above will no longer be in use. I scanned the other libraries the plugin uses and noticed a handful have dropped support for Python2; those libraries are also included via their version to ensure compatability.

Good luck and please let me know if this resolves your issue.

SharpSh00ter109, 2020-12-24:
Installed onto the first system with no issues, finishing installing on the other 8 and then will configure them. So far all is going well though! Thank you for pushing this update. I'll let you know if any other issues come up with this update.

closed 2020-12-30 by SharpSh00ter109

Snortimer, 2020-12-30:
Just reporting in that this has worked perfectly for me for a few days now. Thank you so much for the update.

vsigler, 2020-12-30:
I can also confirm it worked for me, without any Octoprint reinstalls, just installed the new version. Thanks a lot!