fraschetti/Octoslack #114: slack app bot invalid_auth even though it's valid

blissend, 2021-03-06:
I can not get app bot to work. In the logs it shows this...

`octoprint.plugins.Octoslack - ERROR - Error in Slack Web API event loop, Error: invalid_auth`

However if I use the slack documentation demo curl call using exact same token it works...

```
# curl -X POST -H 'Authorization: Bearer <REMOVED>' -H 'Content-type: application/json' --data '{"channel":"octopi","text":"I hope the tour went well, Mr. Wonka.","attachments": [{"text":"Who wins the lifetime supply of chocolate?","fallback":"You could be telling the computer exactly what it can do with a lifetime supply of chocolate.","color":"#3AA3E3","attachment_type":"default","callback_id":"select_simple_1234","actions":[{"name":"winners_list","text":"Who should win?","type":"select","data_source":"users"}]}]}' https://slack.com/api/chat.postMessage
{"ok":true,"channel":"C01QE865N5Q","ts":"1615041757.001500","message":{"bot_id":"B01QSR807R7","type":"message","text":"I hope the tour went well, Mr. Wonka.","user":"U01Q7DC140N","ts":"1615041757.001500","team":"T016QJ0J06L","bot_profile":{"id":"B01QSR807R7","deleted":false,"name":"octopi","updated":1615040031,"app_id":"A01QSEUDX7T","icons":{"image_36":"https:\/\/avatars.slack-edge.com\/2021-03-05\/1842511400513_0c43f86d2a27246f307c_36.png","image_48":"https:\/\/avatars.slack-edge.com\/2021-03-05\/1842511400513_0c43f86d2a27246f307c_48.png","image_72":"https:\/\/avatars.slack-edge.com\/2021-03-05\/1842511400513_0c43f86d2a27246f307c_72.png"},"team_id":"T016QJ0J06L"},"attachments":[{"callback_id":"select_simple_1234","fallback":"You could be telling the computer exactly what it can do with a lifetime supply of chocolate.","text":"Who wins the lifetime supply of chocolate?","id":1,"color":"3AA3E3","actions":[{"id":"1","name":"winners_list","text":"Who should win?","type":"select","data_source":"users"}]}]},"warning":"missing_charset","response_metadata":{"warnings":["missing_charset"]}}
```

So I know the token works and that the bot is in the channel. Perhaps something else is going on?

sfoley, 2021-03-11:
It looks like the Slacker python package is no longer supported, and their auth header doesn't play well with Slack at the moment. I'm new to this project, so maybe this is a recent slack change? I got it to work with a little modification to the Slacker.__init__.py file of line 69 (of the latest version) to change from `kwargs.setdefault('params', {})['token'] = self.token` to `kwargs.setdefault('headers', {})['Authorization'] = "Bearer " + self.token`

Ultimately, the longer term solution may be to move away from using the unsupported Slacker package and use Slack's own SDK.

fraschetti, 2021-03-12:
Hi @sfoley 

I agree - I recall looking into moving away from Slacker a few months ago but I haven't been able to put aside the time to knock this out just yet. A quick hack (which could have some legs since the Slacker project is dead) would be to patch that method but that's never ideal.

TBD when I'll get around to knocking this out but with Slacker long since dead, it's about time I get rid of that dependency.

savage4618, 2021-03-13:
following because i'm having the same issue.

eecue, 2021-03-30:
Following, currently using the web hook method but I really want the full bot!

fraschetti, 2021-04-03:
Update - I've ripped out Slacker and am in the testing phase of Pytyon2 support. Once that testing (and corresponding bug fixes) is complete, I'll add Python3 support, go through that testing cycle and then we should be good to go.

No ETA yet but these changes are underway.

eecue, 2021-04-03:
Awesome! I’m down to pull down a branch and test.

fraschetti, 2021-04-05:
Hi @eecue 

https://github.com/fraschetti/Octoslack/tree/SlackerRemoval

Looking good for Python2, haven't started work on Python3.

eecue, 2021-04-30:
Heya @fraschetti ... any update here? I'm running Python 3 so didn't get a chance to test your PR.

dstrivelli, 2021-05-03:
Same boat. I'm happy to take a look at whatever python3 work you are able to put together.

fraschetti, 2021-05-05:
FYI - I've started work on Python3 support for the official Slack Python library. Stay tuned.

fraschetti, 2021-05-08:
Update - I've completed both Python2 & Python3 development and testing. Changes has been submitted to that development branch:
https://github.com/fraschetti/Octoslack/tree/SlackerRemoval

I'll let these bake for a couple of days but pending any bug reports, a new release with these changes should be coming soon.

fraschetti, 2021-05-08:
Thanks for the feedback @eecue 
Just as an FYI, can you share what settings you had to delete?
I'm looking to see if this is something other users might encounter and I need to code for or if you had something special on your end that I don't need to worry about.

Thanks in advance.

eecue, 2021-05-09:
So I first tried to just check out your branch and copy the files into the python folder over the existing ones. That didn’t work and octoslack didn’t even appear. Then I just downloaded the whole zip of the branch and upload it via the web interface. That started working again and my old settings were in there but I got an error message. I posted it earlier and then edited my message, sorry. I can grab it from the logs once I’m home later. I then removed and cleaned the plug-in, added it again using the zip upload method, added back my creds and all was good.

dstrivelli, 2021-05-11:
@fraschetti I can confirm what eecue was saying. I pulled the plugin and installed via ZIP. I too had to trash creds, but all looks good.

fraschetti, 2021-05-12:
@dstrivelli Did you need to create new creedentials on the Slack side or did you need to reenter your credentials in the Octoslack config UI?

I'd of course like to track down any issues on my side but if there's something I can't fix, I'd ideally likely to call this out  in the release notes.

Thanks again.

eecue, 2021-05-12:
@dstrivelli so I had already created both the bot token and the web hook info and added both to the config, but was using the web hook as the bot didn't work before your awesome update.  so I did not have to regenerate anything, just deleted and paste it back in ... btw here are the errors I saw before I deleted:

> May  8 14:10:54 octopi octoprint[17546]: 2021-05-08 14:10:54,336 - octoprint.plugins.Octoslack - ERROR - Slack Web API - Error processing message: 'user'
> May  8 14:10:54 octopi octoprint[17546]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_Octoslack/__init__.py", line 2181, in retrieve_channel_history

dstrivelli, 2021-05-15:
@fraschetti I did **not** need to recreate the Slack credentials, I just needed to delete OctoSlack plugin from OctoPrint entirely, and all of its data, then install from the ZIP file. I had the same error that @eecue posted above when I had just installed the updated version over the top of the existing install.

```
octoprint.plugins.Octoslack - ERROR - Slack Web API - Error processing message: 'user'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_Octoslack/__init__.py", line 2181, in retrieve_channel_history
    msg_user = message["user"]
KeyError: 'user'
```

But when I specified an authorized user in the plugin settings, everything was good to go. Doesn't seem to work in private channels, but that may be by design or at least a known limitation, I'm not sure. Otherwise, works great.

fraschetti, 2021-05-15:
I found a few extra places where I need to make changes to adapt to the new Slack client response objects (no longer a native dictionary and they don't fully support all dictionary functionality).

I just pushed 2.2.0 - https://github.com/fraschetti/Octoslack/releases/tag/2.2.0

Thanks for your help!