fraschetti/Octoslack #125: Emoji in PushOver not working/supported

TheLion, 2021-09-03:
Hi,

Are emoji's in Pushover not working/supported?

Or is a different syntax required than the `:heavy_minus_sign:` that is used as default in the messages.

Pushover does support emoji's, as the other Pushover plugin does send emoji's, but that doesn't have the customizability that Octoslack offers.

TylonHH, 2022-01-20:
Those syntax is for Slack. Just copy an emoji from somewhere and put it in. for eg https://getemoji.com/

natrius, 2023-08-04:
Why not change the default to emojis? Matrix is using the `:heavy_minus_sign:` as well, but i get `:heavy_minus_sign:` instead of ➖ unfortunately. 
If this could be changed by default to emojis and not to the code, it would be good as it seems it affects more messengers. 

Or, otherwise a setting or a button to switch the text for the fitting emojis? I did it now manually.

TheLion, 2024-05-01:
> Those syntax is for Slack. Just copy an emoji from somewhere and put it in. for eg https://getemoji.com/

This works. Used [https://emojipedia.org](https://emojipedia.org) as the search function worked better for me.

closed 2024-05-01 by TheLion