fraschetti/Octoslack #133: basic authentication throws error

TheLion, 2022-04-10:
Hi,

I secured my MJPEG-STREAMER stream and snapshot with a username/password. This works just fine in Octoprint and in the browser by accessing http://username:password@192.168.1.1:80/?action=snapshot. This syntax is also mentioned within the OctoSlack plugin, but when I use this syntax in Octoprint (which I suppose is used by OctoSlack) it produces an error in the log:

```

2022-04-10 17:49:47,544 - octoprint.plugins.Octoslack - DEBUG - Event: PrintCancelled, NotificationEnabled: True, CommandEnabled: False, OverrideNotificationEnabledCheck: False, LastProcessedKey: PrintCancelled, LastProcessedTime: 0, MinNotificationInterval: 0, Payload: {'name': '5xPiece 1_0.15mm_PLA_Infill_25%_MK3S_4h39m_11.5583mtr.gcode', 'path': '5xPiece 1_0.15mm_PLA_Infill_25%_MK3S_4h39m_11.5583mtr.gcode', 'origin': 'local', 'size': 13922218, 'position': {'z': None, 'x': None, 'e': None, 't': None, 'f': None, 'y': None}, 'owner': 'fourpets', 'user': 'fourpets', 'time': 64.45497759600016, 'reason': 'cancelled', 'slack_rtm_user': None}
2022-04-10 17:49:47,550 - octoprint.plugins.Octoslack - DEBUG - Updating Event last processed time: PrintCancelled, LastProcessedKey: PrintCancelled, NewLastProcessedTime: 1649605787.5505786
2022-04-10 17:49:47,578 - octoprint.plugins.Octoslack - DEBUG - Printer data: {'state': {'text': 'Cancelling', 'flags': {'operational': True, 'printing': True, 'cancelling': True, 'pausing': False, 'resuming': False, 'finishing': False, 'closedOrError': False, 'error': False, 'paused': False, 'ready': False, 'sdReady': True}, 'error': ''}, 'job': {'file': {'name': '5xPiece 1_0.15mm_PLA_Infill_25%_MK3S_4h39m_11.5583mtr.gcode', 'path': '5xPiece 1_0.15mm_PLA_Infill_25%_MK3S_4h39m_11.5583mtr.gcode', 'display': '5xPiece 1_0.15mm_PLA_Infill_25%_MK3S_4h39m_11.5583mtr.gcode', 'origin': 'local', 'size': 13922218, 'date': 1649344417}, 'estimatedPrintTime': 17263.502640947765, 'averagePrintTime': 16777.863726880398, 'lastPrintTime': 16791.44343716299, 'filament': {'tool0': {'length': 10991.865129194473, 'volume': 26.43853397214425}}, 'user': 'fourpets'}, 'currentZ': None, 'progress': {'completion': None, 'filepos': None, 'printTime': None, 'printTimeLeft': None, 'printTimeLeftOrigin': None}, 'offsets': {}, 'resends': {'count': 0, 'transmitted': 30, 'ratio': 0}}
2022-04-10 17:49:47,764 - octoprint.plugins.Octoslack - DEBUG - Octoslack connection method: PUSHOVER
2022-04-10 17:49:47,792 - octoprint.plugins.Octoslack - DEBUG - Upload snapshot - snapshot_upload_method: PUSHOVER
2022-04-10 17:49:47,793 - octoprint.plugins.Octoslack - DEBUG - Local camera settings - Snapshot URL:http://username:password@192.168.1.1:80/?action=snapshot, FlipH: False, FlipV: False, Rotate90: False
2022-04-10 17:49:47,793 - octoprint.plugins.Octoslack - DEBUG - Snapshot URLs: [('http://username:password@192.168.1.1:80/?action=snapshot', False, False, False)]
2022-04-10 17:49:47,817 - octoprint.plugins.Octoslack - ERROR - Error downloading snapshot - URL: http://192.168.1.1:80/?action=snapshot, Error: a bytes-like object is required, not 'str'
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_Octoslack/__init__.py", line 5063, in download_image
    "%s:%s" % (basic_auth_user, basic_auth_pwd)
  File "/usr/lib/python3.7/base64.py", line 58, in b64encode
    encoded = binascii.b2a_base64(s, newline=False)
TypeError: a bytes-like object is required, not 'str'
2022-04-10 17:49:47,831 - octoprint.plugins.Octoslack - DEBUG - Downloaded all 1 snapshots in 0.04 seconds
2022-04-10 17:49:47,831 - octoprint.plugins.Octoslack - DEBUG - download_image thread_responses: [(None, "a bytes-like object is required, not 'str'")]
2022-04-10 17:49:47,831 - octoprint.plugins.Octoslack - DEBUG - postMessage - Channels: $myself$, JSON: [{"mrkdwn_in": ["text", "pretext"], "footer": "Printer: Cancelling, Bed: 50.5\u00b0C/60.0\u00b0C, Nozzle: 161.58\u00b0C/160.0\u00b0C, RasPi: 43.8\u00b0C", "fields": [], "fallback": "Print cancelled: 5xPiece 1_0.15mm_PLA_Infill_25%_MK3S_4h39m_11.5583mtr.gcode", "pretext": "Prusa i3 MK3S+ - Printing cancelled.", "color": "warning", "text": "<b>File</b> 5xPiece 1_0.15mm_PLA_Infill_25%_MK3S_4h39m_11.5583mtr.gcode (13.3MB via OctoPrint)\n<b>User</b> fourpets\n<b>Snapshot error(s):</b>\n <b> - </b> a bytes-like object is required, not 'str'"}]
2022-04-10 17:49:47,831 - octoprint.plugins.Octoslack - DEBUG - Send Pushover msg start
2022-04-10 17:49:47,854 - octoprint.plugins.Octoslack - DEBUG - Pushover msg title: Prusa i3 MK3S+ - Printing cancelled.
2022-04-10 17:49:47,857 - octoprint.plugins.Octoslack - DEBUG - Pushover msg body: <b>File</b> 5xPiece 1_0.15mm_PLA_Infill_25%_MK3S_4h39m_11.5583mtr.gcode (13.3MB via OctoPrint)
<b>User</b> fourpets
<b>Snapshot error(s):</b>
 <b> - </b> a bytes-like object is required, not 'str'

```

Not quite sure why this happens

FrugalGuy, 2022-08-12:
Message means you supplied a string argument when a bytes argument was required. 
Use 'theArgument'.encode('utf-8') to convert the string 'theArgument' into bytes b"theArgument"
Which argument? Check the docs for the failing API call.

TheLion, 2022-08-12:
> @TheLion Message means you supplied a string argument when a bytes argument was required. Use 'theArgument'.encode('utf-8') to convert the string 'theArgument' into bytes b"theArgument" Which argument? Check the docs for the failing API call.

I understand the error, but to be clear, I didn't code anything. I just entered the URL I mentioned in the box, that's it.

fraschetti, 2022-11-04:
Thanks for reporting this @TheLion - I'll see if I can repro this setup.
Having not yet looked myself, if you have a link to the doc you followed to lock down your video stream, please pass it along as it might save me a bit of time and effort.