fraschetti/Octoslack #135: Unable to pass variables from Octoprint/Octoslack to a Slack Webhook

MrGeekFreak, 2022-07-30:
Hello all!

I am trying to integrate Octoslack into a Slack Workspace that does not have bots/custom applications available. I was experimenting with a Workflow Webhook, but can't seem to pass variables from Octoslack to it. I can get Octoprint to post a generic message in a Slack board whenever the printer performs GCODE. Any time I add a variable to the Slack Webhook, I begin getting HTTP error 400 due to Octoslack using different variables.

What variables do I add to my Slack Webhook (included screenshot) to make this work?
[image]

fraschetti, 2022-11-04:
Hi @MrGeekFreak 

I'm not sure if you're still having this issue or if you even still care but I'll attempt to reproduce this in the near-ish future and will report back when I have something to share.

MrGeekFreak, 2022-11-13:
Hey @fraschetti !
> I'm not sure if you're still having this issue or if you even still care but I'll attempt to reproduce this in the near-ish future and will report back when I have something to share.

That would be super appreciated! I am definitely still interested in getting OctoSlack working with a Slack Workflow/WebHook URL.

Thank you!

MrGeekFreak, 2022-11-13:
Oops - I did not mean to close this out yet.

MrGeekFreak, 2022-11-14:
`2022-11-13 18:24:28,946 - octoprint.plugins.Octoslack - ERROR - Slack WebHook message send failed: {"error":"missing_field_value","ok":false,"response_metadata":{"messages":["[ERROR] empty required field: 'messages'"]}}`

The OctoPrint log file is giving me this error when attempting to send a notification through Slack.