fraschetti/Octoslack #136: FYI: Octopi moves location of vcgencmd

ChrisHeerschap, 2022-08-24:
Running OctoPi 1.0.0 RC1 and OctoSlack wasn't working. Found the cause in the logs:

```
FileNotFoundError: [Errno 2] No such file or directory: '/opt/vc/bin/vcgencmd'
```

Investigated and saw that the underlying OS for this release doesn't put vcgencmd in that folder anymore, it's now located in `/bin/vcgencmd`

I looked through the configs and didn't see where that path was coded so couldn't propose a change, but there is a simple update:

```
sudo mkdir -p /opt/vc/bin
sudo ln -s /bin/vcgencmd /opt/vc/bin
```

fraschetti, 2022-11-04:
Received. Thanks for reporting and sharing the updated path.

fraschetti, 2022-11-05:
I've pushed a change to ideally transparently & efficiently support the new & old paths.
Thanks for reporting this issue, the change will of course be available in the next release.